board707/DMD_STM32 #122: Connection question on Parallel Monochrome example

xuniuer, 2024-08-12:
Hi there,

I now test library with RP2040 PICO to drive P10 monochrome panel. Here is a question about `row1, row2` as follows -- `7, 8`:

```
#if defined(ARDUINO_ARCH_RP2040)
#if defined(DMD_PARA)
#define DMD_PIN_A 1
#define DMD_PIN_B 2
#define DMD_PIN_nOE 14
#define DMD_PIN_SCLK 15

// These pins must be consecutive in ascending order
uint8_t pins[] = { 6, 7, 8 };  // CLK , row1, row 2
#else
#error Monochrome_SPI mode unsupported for Rasberry Pico RP2040
#endif
#endif
```

As the example says, it's for `ACROSS 1, DOWN 2` connection of two parallel panels. How do you make the connection which I could not figure out?

Thanks in advance.

board707, 2024-08-12:
Hi
Sorry, I don't understand, what is your issue...

xuniuer, 2024-08-12:
Hi, thanks for your attention. In fact I want to connect 14 panels (P10, monochrome) together to gen a big matrix. I know the library is already flexible to support my requirements, such as connection schema and `DMD_Monochrome_Parallel` . I have checked your wiki to confirm it. Here just want to confirm cable connection for example of `dmd_monochrome` where two panels are paralleled. If I should connect `DMD_PIN_A, DMD_PIN_B, OE, SCLK` and `CLK` to panel#1 and panel#2 at the same time, then, separately, connect `6 - row1` to panel#1, connect `7 - row2` to panel#2, just defined in example source code.

---- Replied Message ----
| From | ***@***.***> |
| Date | 08/12/2024 15:23 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [board707/DMD_STM32] Connection question on Parallel Monochrome example (Issue #122) |

Hi
Sorry, I don't understand, what is your issue...

board707, 2024-08-12:
> I want to connect 14 panels

What is configuration of your big screen? Is it 7 panels wide and 2 rows high?

xuniuer, 2024-08-22:
Sorry for late to reply because of frequently broken connection to github from China mainland.

My screen is as big as your guess. I have tested it by using one chain on RP2040 pico. Thanks.

Plus, I wonder if the library could support a true color screen with 240x160 resolution?

xuniuer, 2024-08-22:
Sorry for late to reply because of frequently broken connection to github from China mainland.

My screen is as big as your guess. I have tested it by using one chain on RP2040 pico. Thanks.

Plus, I wonder if the library could support a RGB screen with 240x160 resolution?

board707, 2024-08-22:
> I have tested it by using one chain on RP2040 pico. Thanks.

Hope you solved the problems with parallel connections.

> I wonder if the library could support a RGB screen with 240x160 resolution?

No, that would be too good :) The tested maximum is 256x64. I planned some improvements, but I don't think it will be more than 256x128 in nearest feature.

xuniuer, 2024-08-23:
Thank you!

> I planned some improvements, but I don't think it will be more than 256x128 in nearest feature.

Sounds great already! I will have more tries with the library.

closed 2024-08-23 by xuniuer