board707/DMD_STM32 #123: Add a new panel

Vitdesp, 2024-08-15:
## Panel description 
#### Dimensions (in pixels) : 128x64
#### Scan factor: 1/32
#### Chips on rear side:  FM6126B, DP32020A

## Panel photos

## Sample videos

## Specifications

  - Library version: 1.1.4
  - Platform: STM32
  - MCU board:  STM32F103C8
  - Connections: default pins
Good day. They gave me a panel with a set of FM6126B and DP32020A. On DMD_RGB_FM6126 the panel does not light up. On DMD_RGB_SHIFTREG_ABC it works but lights up very dimly. By 30-40 percent. Help increase the brightness.

board707, 2024-08-15:
Hi
Please show your test code using DMD_RGB_FM6126  class.

Vitdesp, 2024-08-16:
`#include "DMD_RGB_FM6126a.h"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB3
#define DMD_PIN_B PB6
#define DMD_PIN_C PB4
//#define DMD_PIN_D PB3
//#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C }; //, DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS> 
// <COLOR_1BITS> mode is not recomended for bitmap

//DMD_RGB <RGB32x16plainS4, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB_FM6126 <RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    // uncomment the line below if you using St-Link with STM32F103xx boards
    //enableDebugPorts();
    
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(250);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}`

board707, 2024-08-16:
please try this combination of DMD_RGB_FM6126 and DMD_RGB_SHIFTREG_ABC classes:

```
#include "DMD_RGB_FM6126a.h"

template <int MUX_CNT, int P_Width, int P_Height, int SCAN, int SCAN_TYPE, int COL_DEPTH>
class DMD_RGB_FM6126_SR : public DMD_RGB_FM6126<MUX_CNT, P_Width, P_Height, SCAN, SCAN_TYPE, COL_DEPTH>
  {

  public:
    DMD_RGB_FM6126_SR(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
      byte panelsWide, byte panelsHigh, bool d_buf = false) :
      DMD_RGB_FM6126<MUX_CNT, P_Width, P_Height, SCAN, SCAN_TYPE, COL_DEPTH>
      (mux_list, _pin_nOE, _pin_SCLK, pinlist, panelsWide, panelsHigh, d_buf)
      {}

protected:
  void generate_muxmask() override
    {
    pinMode(this->mux_pins[0], OUTPUT);
      pinMode(this->mux_pins[1], OUTPUT);
      pinMode(this->mux_pins[2], OUTPUT);
    }

  void set_mux(uint8_t curr_row) override {
    byte pin_DMD_A = this->mux_pins[0];
    byte pin_DMD_B = this->mux_pins[1];
    byte pin_DMD_C = this->mux_pins[2];
    // Just shift the row mux by one for incremental access
    digitalWrite(pin_DMD_B, HIGH);
    digitalWrite(pin_DMD_C, (curr_row == 0)); // Shift out 1 for line 0, 0 otherwise
    digitalWrite(pin_DMD_A, HIGH); // Clock out this bit
    digitalWrite(pin_DMD_A, LOW);
    digitalWrite(pin_DMD_B, LOW);
  }

};
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB3
#define DMD_PIN_B PB6
#define DMD_PIN_C PB4
//#define DMD_PIN_D PB3
//#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C }; //, DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4 - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS>
// <COLOR_1BITS> mode is not recomended for bitmap

DMD_RGB_FM6126_SR <RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0; // background - black
uint16_t fg = 0; // foreground

void setup(void)
{
// uncomment the line below if you using St-Link with STM32F103xx boards

enableDebugPorts();

dmd.init();
fg = dmd.Color888(0, 0, 255);
dmd.setBrightness(250);

}

void loop(void) {
// fill the matrix with points row by row
for (int i = 0; i < dmd.height(); i++) {
for (int j = 0; j < dmd.width(); j++) {
dmd.drawPixel(j,i, fg);
delay(30);
}
}
// clear the screen
dmd.fillScreen(bg);
}
```

Vitdesp, 2024-08-16:
Everything is broken. There was version 0.7.0. DMD_RGB_SHIFTREG_ABC worked very poorly.
For DMD_RGB_FM6126 to work, I installed 1.1.4. Updated Hardware: https://github.com/board707/Arduino_STM32/tree/lto_for_c6
Now nothing is displayed with DMD_RGB_SHIFTREG_ABC and with the example you suggested.
What should I do?

board707, 2024-08-16:
Hi
I just tested the library version 1.1.4 with my FM6126 panel and have to admit that this version has problems with the 6126 driver. I don't know what's wrong yet, but I'll try to figure it out in the next few days.

Vitdesp, 2024-08-17:
OK

board707, 2024-08-20:
Hi
Please test the code I suggested above with 1.1.0 release
https://github.com/board707/DMD_STM32/releases/tag/v1.1.0

Vitdesp, 2024-08-21:
[image]

The same.

board707, 2024-08-21:
Please to be more clear.
What is "the same" ?
Why did you show me this picture?

board707, 2024-08-22:
Hi
The bug with FM6126 panels not works in latest versions was fixed. You can update your copy of the library by knob "Download ZIP" on the library's Github (do not download the releases - it not updated yet).

Vitdesp, 2024-09-08:
Good day.
Didn't have time to answer. Got sick.
Sent a photo just to show the location of the chips.
1.1.4 and 1.1.5 did not give a result. Nothing is displayed on the panel.
The signal on the OE and LAT pins is the same. There is nothing on the CLK pin. There are line selection signals on the D and E pins. But on my panel, pins D and E are connected to ground.
Maybe the problem is different.
I installed many versions of libraries, various add-ons. And I'm already confused.
I want to remove everything and install a clean one.
Arduino 1.8.19.
Can you give a link to the Clark add-on that you are working with now. and maybe I missed something else.
Maybe Adafruit GFX is not the right one.
I will check everything.

board707, 2024-09-09:
Hi
Please show your latest code, that you used to work with a 1.1.5 version of the library.

Vitdesp, 2024-09-09:
Hello.
I tried the example you suggested. The screen is black.
Example dmd rgb Pattern test. The screen is black.
There is no CLK signal on the Hub75.
I changed COLOR_4BITS to COLOR_4BITS_Packed and the CLK signal appeared.
Lines started to be drawn. BUT double, 16 at a time.
The first pass draws 1,2, 5,6, 9,10, 13,14, 17,18, 21,22, 25,26, 29,30 lines.
The second pass draws 2,3, 6,7, 10,11, etc.
The third pass repeats the second, but with greater brightness.
The fourth pass draws 3,4, 7,8, 11,12, etc.
The fifth pass repeats the fourth, but with greater brightness.
At first the brightness is about 1/4, with each pass it increases.
Passage 33. The same thing starts at the bottom of the screen.

Vitdesp, 2024-09-09:
[image]

Vitdesp, 2024-09-09:
[image]

Vitdesp, 2024-09-09:
Couldn't find a datasheet for DP32020A. 
I roughly understood that the data goes sequentially to pin 2, 
Clock to pin 3.
5,6,7,8,13,14,15,16 outputs.

board707, 2024-09-10:
> Couldn't find a datasheet for DP32020A.

[DP32020A_REV1.3_EN.zip](https://github.com/user-attachments/files/16940126/DP32020A_REV1.3_EN.zip)

Vitdesp, 2024-09-10:
Hello. Thank you.
Everything works. But the brightness is low, even when set to maximum. Can the brightness be increased further?

board707, 2024-09-10:
> does not work with COLOR_4BITS. But with COLOR_4BITS_Packed it works.

The pins for this two modes are different. 
Use this for  COLOR_4BITS mode:
```
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1
```

Vitdesp, 2024-09-11:
Hello.
I can't use PA6, the board is laid out on PA15 as it was before. I can't redesign the board.

board707, 2024-09-12:
> I can't use PA6,

In this case, you will not be able to use `COLOR_4BITS` mode. 
These two modes do the same thing, only the `PACKED` one uses memory more economically, and `COLOR_4BITS` is slightly faster. Therefore, the first is recommended for the STM32F1 series, and the second for the STM32F4. Otherwise, they replace each other, so there is no problem if only one of them works on your board.

Vitdesp, 2024-09-14:
Hello.
Since the brightness is very low, I tried to use the COLOR_4BITS mode,
PA6 and PA15. There are no signals on all pins.
Only COLOR_4BITS_Packed works.
Help increase the brightness.

Vitdesp, 2024-09-16:
Good day.
I want to apologize for the brightness. It was a panel malfunction. 74HC245 is poorly soldered. Now the brightness is excellent.
But when I turn on the COLOR_4BITS mode, there are no signals on all the pins.
If I turn on the COLOR_4BITS_Packed mode, the signals appear.

Vitdesp, 2024-09-16:
/[code]
#include "DMD_RGB_FM6126a.h"

template <int MUX_CNT, int P_Width, int P_Height, int SCAN, int SCAN_TYPE, int COL_DEPTH>
class DMD_RGB_FM6126_SR : public DMD_RGB_FM6126<MUX_CNT, P_Width, P_Height, SCAN, SCAN_TYPE, COL_DEPTH>
{

  public:
    DMD_RGB_FM6126_SR(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
                      byte panelsWide, byte panelsHigh, bool d_buf = false) :
      DMD_RGB_FM6126<MUX_CNT, P_Width, P_Height, SCAN, SCAN_TYPE, COL_DEPTH>
      (mux_list, _pin_nOE, _pin_SCLK, pinlist, panelsWide, panelsHigh, d_buf)
    {}

  protected:
    void generate_muxmask() override
    {
      pinMode(this->mux_pins[0], OUTPUT);
      pinMode(this->mux_pins[1], OUTPUT);
      pinMode(this->mux_pins[2], OUTPUT);
    }

    void set_mux(uint8_t curr_row) override {
      byte pin_DMD_A = this->mux_pins[0];
      byte pin_DMD_B = this->mux_pins[1];
      byte pin_DMD_C = this->mux_pins[2];
      // Just shift the row mux by one for incremental access
      digitalWrite(pin_DMD_B, HIGH);
      digitalWrite(pin_DMD_C, (curr_row == 0)); // Shift out 1 for line 0, 0 otherwise
      digitalWrite(pin_DMD_A, HIGH); // Clock out this bit
      digitalWrite(pin_DMD_A, LOW);
      digitalWrite(pin_DMD_B, LOW);
    }

};
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB3
#define DMD_PIN_B PB6
#define DMD_PIN_C PB4
//#define DMD_PIN_D PB3
//#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C }; //, DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0, PA1, PA2, PA3, PA4, PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4 - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS>
// <COLOR_1BITS> mode is not recomended for bitmap

DMD_RGB_FM6126_SR <RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0; // background - black
uint16_t fg = 0; // foreground

void setup(void)
{
  // uncomment the line below if you using St-Link with STM32F103xx boards

  enableDebugPorts();

  dmd.init();
  dmd.setBrightness(250);
  
  dmd.fillScreen(dmd.Color888(255, 0, 0));
  delay(1000);
  dmd.fillScreen(dmd.Color888(0, 255, 0));
  delay(1000);
  dmd.fillScreen(dmd.Color888(0, 0, 255));
  delay(1000);
  dmd.fillScreen(dmd.Color888(255, 255, 255));
  delay(1000);
  
  fg = dmd.Color888(0, 0, 250);
}

void loop(void) {
  // fill the matrix with points row by row
  for (int i = 0; i < dmd.height(); i++) {
    for (int j = 0; j < dmd.width(); j++) {
      dmd.drawPixel(j, i, fg);
      delay(30);
    }
  }
  // clear the screen
  dmd.fillScreen(bg);
}
[/code]
/

Vitdesp, 2024-09-18:
Good day.
The outputs D, E have a high level. And on the panel, pins D, E are connected to ground. The CPU is very hot. How can I give a low level to these outputs?

board707, 2024-09-18:
> The outputs D, E have a high level.

What are the outputs D and E? 
In the code above you didn't define any pins as D & E signals:
```
#define DMD_PIN_A PB3
#define DMD_PIN_B PB6
#define DMD_PIN_C PB4
//#define DMD_PIN_D PB3
//#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C };
```
You commented out the DMD_PIN_D & E and not included it in the list - so I don't understand, on which pins did you see a high level.

Vitdesp, 2024-10-31:
And if I may, another question off-topic.
The library description says that in monochrome mode you can connect up to 100 panels. And RGB up to 16 64x32.
But it is not specified when using which controller.
And is it possible to combine 2 regular 128x64 RGB panels on STM32F103C8 or do you need another controller?

board707, 2024-11-01:
> The library description says that in monochrome mode you can connect up to 100 panels. And RGB up to 16 64x32.

For monochrome, RP2040 is assumed in parallel mode. For RGB, this is also RP2040

> is it possible to combine 2 regular 128x64 RGB panels on STM32F103C8

One 128x64 panel needs a 16 Kbyte buffer in 4bit color mode, so it is not possible to control two panels with  STM32F103C8, having 20K RAM only

Vitdesp, 2024-11-01:
Thank you. Got it.

closed 2024-11-09 by board707