board707/DMD_STM32 #127: Add a new panel 128x64 s32 DP3364 driver

funnymind, 2024-08-30:
## Panel description 
#### Dimensions (in pixels) : 128x64
#### Scan factor: 1/32
#### Chips on rear side:  DP3364, 74HC245KA

## Panel photos
1.
[image]
2.
[image]
3.
[image]
4.
[image]

## Sample videos

## Specifications

  - Library version: 1.1.5
  - Platform: STM32
  - MCU board: blackpill stm32f411ceu6
  - Connections: used pins A=PB6, B=PB5, C=PB4, D= PB3, nOE=PB0, SCLK=PB7, CLK=PA6, R0=PA0, G0=PA1, B0=PA2,
R1=PA3, G1=PA4, B1=PA5

funnymind, 2024-08-30:
Hi @board707.
I have similar panels as were in this request #91. I saw that you were ready to do some tests on them, if you still don't mind then I'm ready.

board707, 2024-08-30:
Hi @funnymind 
Thank you for your kind offer of help.

The driver DP3364 is a S-PWM type. In order for it to work, it is necessary to do some reverse engineering to figure out it protocol and configuration settings.
Honesty speaking, it is very difficult to test unknown driver remotely. Fortunately, one library friend presents me a two panels with DP3364 driver :), so now I can work with them directly.
The issue is that I haven't much free time last days, so I can't say exactly when i'll be able to work on these panels.

closed 2024-12-12 by board707

martodg, 2025-11-07:
Hi @board707 
I purchased a panel from Aliexpress with the DP3364S chip. If you ever update the library to add this driver, I will be very grateful.
https://www.aliexpress.com/item/1005006086019360.html
Thanks.