board707/DMD_STM32 #128: Issue with panel 64x64 chip FM6124

impuls-op, 2024-09-04:
Indoor Panel 64x64 with chip FM6124
Used sketch: example/dmd_rgb.ino
In wiki the chip FM6124 marked as Tested with DMD_STM32 library.
Maybe something needs to be changed in the config? 

[image]

https://github.com/user-attachments/assets/4eefce6e-86f3-42c4-81dd-6eda272ffe15

`/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32 and RP2040 boards
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false
/* =================== *
*     STM32F4 pins     *
*  =================== */
#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 10
#define DMD_PIN_D 13
#define DMD_PIN_E 14
[... 174 lines cut ...]
                    }
                    else {
                        if (step[i] < 0) dmd.drawMarqueeX(m, 0, dmd.height());
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

                // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

         
        }
    }
}

`

board707, 2024-09-04:
Hi
What board do you use to test the panel?

impuls-op, 2024-09-04:
> Hi
> What board do you use to test the panel?

Ri Pico 2040

board707, 2024-09-04:
Hmm...
The pins that you used for A B C D E signals are incorrect.
Please read the comment above pin list in the code: 
```
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 10
#define DMD_PIN_D 13
#define DMD_PIN_E 14
```

impuls-op, 2024-09-04:
Its worked! Now used: 
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10

But i have many artefacts

https://github.com/user-attachments/assets/4c53017f-9831-40ad-92c6-e84e1616548a

board707, 2024-09-04:
Try checking the contacts. It looks more like bad connections or some kind of interference.

impuls-op, 2024-09-08:
Artifacts ended when I lowered the power supply to 4.5v for information

closed 2024-09-09 by board707