board707/DMD_STM32 #130: Flex panels support

DoomHammer, 2024-09-22:
## Panel description 
#### Dimensions (in pixels) : 64x32
#### Scan factor: 1/16

## Panel photos

[image]
[image]
[image]

## Specifications

  - Library version: 1.01
  - Platform: Pi Pico
  - MCU board: Pi Pico
  - Connections: used pins

I have some flexible panels. Some of them are 64x32, some are 128x64. Whatever I do I cannot get them to display anything. Any clue what can I do to make them work?

board707, 2024-09-23:
Hi

As I mentioned previously, the chip DP3265 is a S-PWM type driver. This driver is not supported in the current version of the library.

DoomHammer, 2024-09-23:
What's necessary to add this support?

board707, 2024-09-23:
The two things :) - i) I need a panel with the driver to be in my hands, ii) I need a lot of free time for testing. And the biggest problem with the latter.

DoomHammer, 2024-09-23:
FIrst part is doable as I have a few of those. For the latter, I'm afraid I can't help. There's no rush with that, though.