board707/DMD_STM32 #131: DISPLAYS_ACROSS >3 not working

ravigyanvihar, 2024-10-02:
sir i am trying to make a panel with 4x5 with 80X40 led p6 matrix on raspberry pi pico 
this is my settings
#define DISPLAYS_ACROSS 4
#define DISPLAYS_DOWN 5
DMD_RGB_SHIFTREG_ABC<RGB80x40plainS20, COLOR_1BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

but when I set DISPLAYS_ACROSS 3, it works but if I set DISPLAYS_ACROSS 4 the logic stops working.

can you please help me with this

board707, 2024-10-02:
Hi
I think the problem is not DISPLAYS_ACROSS, but the total display size. The maximum screen size tested with Raspberry Pico is 16 panels 64x32.
You are trying to run 20 panels 80x40 - that's exactly twice as much. 
Try to set, say, DISPLAYS_DOWN 3 and test DISPLAYS_ACROSS 4

ravigyanvihar, 2024-10-02:
HI 
i HAVE TESTED 24 80X40 DISPLAY ON ONE PICO.
IN  
DISPLAYS_ACROSS 3
DISPLAYS_down 8

BUT EVEN IF I SET 
DISPLAYS_ACROSS 4
DISPLAYS_DOWN 1
IT DOESNT WORK

board707, 2024-10-02:
Ok, thank you.
Please clarify, what is your "DOESNT WORK" means exactly? 
Do you see anything on panels?

ravigyanvihar, 2024-10-02:
//pico version 3.7.2
#include "DMD_RGB.h"
#include <ArduinoJson.h>

#define DISPLAYS_ACROSS 4
#define DISPLAYS_DOWN 2

#define ENABLE_DUAL_BUFFER false

#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10

uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D, DMD_PIN_E };

#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

uint8_t custom_rgbpins[] = { 11, 0, 1, 2, 3, 4, 5 };

DMD_RGB_SHIFTREG_ABC<RGB80x40plainS20, COLOR_1BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t red = dmd.Color888(255, 0, 0);
uint16_t green = dmd.Color888(0, 255, 0);
uint16_t blue = dmd.Color888(0, 0, 255);
uint16_t yellow = dmd.Color888(255, 255, 0);
uint16_t purple = dmd.Color888(255, 0, 255);
uint16_t cyan = dmd.Color888(0, 255, 255);
uint16_t white = dmd.Color888(255, 255, 255);
uint16_t black = dmd.Color888(0, 0, 0);
//DMD_Standard_Font Arial_F(Arial_14);

void main_display() {
  //clear the screen first

  dmd.fillScreen(black);

  //making all border
  //dmd.drawRect(x0,y0, w,h,color);h lin e (xstart,ystart,legth)
  dmd.drawRect(0, 0, dmd.width(), dmd.height(), white);
  dmd.drawPixel(1, 1, white);
  dmd.drawPixel(dmd.width() -2, 1, red);
  dmd.drawPixel(dmd.width() -2, dmd.height() - 2, green);
  dmd.drawPixel(1, dmd.height() - 2, blue);

  //print after update
 // dmd.swapBuffers(true);
}

void setup() {
  // put your setup code here, to run once:
  Serial1.setRX(17u);
  Serial1.setTX(16u);
  Serial1.begin(115200);
  //dmd.setConnectScheme(2);
  dmd.init(100);
  main_display();
  pinMode(25, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(25, HIGH);
  delay(1000);
  digitalWrite(25, LOW);
  delay(1000);
}

it's is my code 
the  code seems stuck .
the led on pin 25 doesn't blink .
and no display on the led matrix

board707, 2024-10-02:
I need make some tests.
What the library version do you use?

ravigyanvihar, 2024-10-02:
i am using 1.1.2

board707, 2024-10-02:
Thank you, I will investigate it

ravigyanvihar, 2024-10-04:
Sir ,
Have you got time to check the issue

board707, 2024-10-04:
May be on weekend.

board707, 2024-10-05:
Hi 
Thank you for your feedback. 
It turned out to be an error in the dimensionality of a variable that overflowed when the screen width was greater than 255 pixels.
I will post a patch when I find all the related errors.

board707, 2024-10-05:
I hope issue fixed. 
Please update the files DMD_RGB.cpp & .h and test your code.

ravigyanvihar, 2024-10-06:
Ok sir ,
I will test and update

ravigyanvihar, 2024-10-12:
hi sir ,
I tried with the update .

display across 4
display down 5

if I set dmd.setCnnectScheme(0);
the display does to turn on. I mean entire code is stuck.

but if I set  dmd.setCnnectScheme(2); or  dmd.setCnnectScheme(1);
the display does boot and works almost fine but the following problem comes  

https://github.com/user-attachments/assets/0f67daf0-0ee9-4e68-956f-5388833195cb

//pico version 3.7.2
#include "DMD_RGB.h"
#include <ArduinoJson.h>

#define DISPLAYS_ACROSS 4
#define DISPLAYS_DOWN 5

//total lines 40x5 = 200
//total paramets = heading + heading of 12 machine display
//200/14 = 14 pixle;

#define ENABLE_DUAL_BUFFER true
#include "Fonts/Arimo_Bold_14.h"
#include "Fonts/null/Serif_plain_12.h"
#include "Fonts/DSEG7_Classic_Bold_25.h"
#include "Fonts/Dialog_plain_25.h"
#include "Fonts/null/Serif_plain_10.h"
#include "Fonts/Serif_bold_20.h"
#include "Fonts/null/Cousine_Regular_8.h"

DMD_GFX_Font Arimo_Bold_14_font((uint8_t*)&Arimo_Bold_14, (uint8_t*)&Arimo_Bold_14Glyphs, 0x80, 13);
DMD_GFX_Font Serif_plain_12_null((uint8_t*)&Serif_plain_12, (uint8_t*)&Serif_plain_12Glyphs, 0x80, 13);
DMD_GFX_Font DSEG7_Classic_Bold_25_Bold((uint8_t*)&DSEG7_Classic_Bold_25, (uint8_t*)&DSEG7_Classic_Bold_25Glyphs, 0x80, 13);
DMD_GFX_Font Dialog_plain_25_Bold((uint8_t*)&Dialog_plain_25, (uint8_t*)&Dialog_plain_25Glyphs, 0x80, 13);
DMD_GFX_Font Serif_plain_10_null((uint8_t*)&Serif_plain_10, (uint8_t*)&Serif_plain_10Glyphs, 0x80, 13);
DMD_GFX_Font Serif_bold_20_null((uint8_t*)&Serif_bold_20, (uint8_t*)&Serif_bold_20Glyphs, 0x80, 13);
DMD_GFX_Font Cousine_Regular_8_null((uint8_t*)&Cousine_Regular_8, (uint8_t*)&Cousine_Regular_8Glyphs, 0x80, 13);

#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10

uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D, DMD_PIN_E };

#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

uint8_t custom_rgbpins[] = { 11, 0, 1, 2, 3, 4, 5 };

DMD_RGB_SHIFTREG_ABC<RGB80x40plainS20, COLOR_1BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t red = dmd.Color888(255, 0, 0);
uint16_t green = dmd.Color888(0, 255, 0);
uint16_t blue = dmd.Color888(0, 0, 255);
uint16_t yellow = dmd.Color888(255, 255, 0);
uint16_t urplep = dmd.Color888(255, 0, 255);
uint16_t cyan = dmd.Color888(0, 255, 255);
uint16_t white = dmd.Color888(255, 255, 255);
uint16_t black = dmd.Color888(0, 0, 0);
//DMD_Standard_Font Arial_F(Arial_14);

int text_color = red;

void display() {
  // maiking all all lines
  int x = 0;
  int y = 0;
  int xMax = 320;
  int yMax = 400;

  for (y = 19; y < yMax; y++) {
    for (x = 0; x < xMax; x++) {
      delay(10);
      dmd.fillScreen(0);
      dmd.selectFont(&Arimo_Bold_14_font);
      String s = "2.0";
      dmd.drawString(x, y, s.c_str(), s.length(), white, 0);
      dmd.swapBuffers(true);
    }
    y = y + 19;
  }

}

void setup() {
  // put your setup code here, to run once:
  // Serial1.setRX(17u);
  // Serial1.setTX(16u);
  // Serial1.begin(115200);
  dmd.setRotation(2);
  dmd.setConnectScheme(2);
  dmd.init(100);
  main_display();
  delay(10000);
  display();
  pinMode(25, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(25, HIGH);
  delay(1000);
  digitalWrite(25, LOW);
  delay(1000);
}

board707, 2024-10-12:
> the following problem comes

I don't understand where the problem, please point it for me

ravigyanvihar, 2024-10-13:
ok
sir 
please check the attached video.

https://github.com/user-attachments/assets/9f171c23-ff50-4d8d-b613-b71c696a0666

when I try to print red color half display becomes red . 
remaining all color are working.

code is as follows 

//pico version 3.7.2
#include "DMD_RGB.h"
#include <ArduinoJson.h>

#define DISPLAYS_ACROSS 4
#define DISPLAYS_DOWN 5

#define ENABLE_DUAL_BUFFER true
#include "Fonts/Arimo_Bold_14.h"
#include "Fonts/null/Serif_plain_12.h"
#include "Fonts/DSEG7_Classic_Bold_25.h"
#include "Fonts/Dialog_plain_25.h"
#include "Fonts/null/Serif_plain_10.h"
#include "Fonts/Serif_bold_20.h"
#include "Fonts/null/Cousine_Regular_8.h"

DMD_GFX_Font Arimo_Bold_14_font((uint8_t*)&Arimo_Bold_14, (uint8_t*)&Arimo_Bold_14Glyphs, 0x80, 13);
DMD_GFX_Font Serif_plain_12_null((uint8_t*)&Serif_plain_12, (uint8_t*)&Serif_plain_12Glyphs, 0x80, 13);
DMD_GFX_Font DSEG7_Classic_Bold_25_Bold((uint8_t*)&DSEG7_Classic_Bold_25, (uint8_t*)&DSEG7_Classic_Bold_25Glyphs, 0x80, 13);
DMD_GFX_Font Dialog_plain_25_Bold((uint8_t*)&Dialog_plain_25, (uint8_t*)&Dialog_plain_25Glyphs, 0x80, 13);
DMD_GFX_Font Serif_plain_10_null((uint8_t*)&Serif_plain_10, (uint8_t*)&Serif_plain_10Glyphs, 0x80, 13);
DMD_GFX_Font Serif_bold_20_null((uint8_t*)&Serif_bold_20, (uint8_t*)&Serif_bold_20Glyphs, 0x80, 13);
DMD_GFX_Font Cousine_Regular_8_null((uint8_t*)&Cousine_Regular_8, (uint8_t*)&Cousine_Regular_8Glyphs, 0x80, 13);

#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10

uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D, DMD_PIN_E };

#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

uint8_t custom_rgbpins[] = { 11, 0, 1, 2, 3, 4, 5 };

DMD_RGB_SHIFTREG_ABC<RGB80x40plainS20, COLOR_1BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t red = dmd.Color888(255, 0, 0);
uint16_t green = dmd.Color888(0, 255, 0);
uint16_t blue = dmd.Color888(0, 0, 255);
uint16_t yellow = dmd.Color888(255, 255, 0);
uint16_t urplep = dmd.Color888(255, 0, 255);
uint16_t cyan = dmd.Color888(0, 255, 255);
uint16_t white = dmd.Color888(255, 255, 255);
uint16_t black = dmd.Color888(0, 0, 0);
//DMD_Standard_Font Arial_F(Arial_14);

int text_color = red;

void display(int col) {
  // maiking all all lines
  int x = 0;
  int y = 0;
  int xMax = 320;
  int yMax = 400;
  //ok posion
  dmd.fillScreen(0);
  dmd.selectFont(&Arimo_Bold_14_font);
  String s = "TEST POS 1";
  dmd.drawString(0, 20, s.c_str(), s.length(), col, 0);
  dmd.swapBuffers(true);
  delay(5000);

  //ok posion
  dmd.fillScreen(0);
  dmd.selectFont(&Arimo_Bold_14_font);
  s = "TEST POS 2";
  dmd.drawString(0, 40, s.c_str(), s.length(), col, 0);
  dmd.swapBuffers(true);
  delay(5000);

}

void setup() {

  dmd.setRotation(2);
  dmd.setConnectScheme(2);
  dmd.init();
  delay(1000);
  pinMode(25, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(25, HIGH);
  delay(1000);
  digitalWrite(25, LOW);
  delay(1000);
  display(white);
  display(red);
  display(green);
  display(blue);
}

board707, 2024-10-13:
Thanks, now I understood.
I have nothing to say just now, I need to run it on my board.

board707, 2024-10-13:
Hi
I couldn't reproduce your results. In my tests everything works fine up to a screen size of 320x224 pixels (5x7 panels 64x32). I tried all screen orientations and connection schemes.

As for your first test - you made a mistake there and are trying to display a string beyond the screen boundaries - your screen is 4x5 panels 80x40 - that is 320x200 pixels, and for some reason you set the Y maximum to 400:
```
void display() {
// maiking all all lines
int x = 0;
int y = 0;
int xMax = 320;
int yMax = 400;

for (y = 19; y < yMax; y++) {
 for (x = 0; x < xMax; x++) {
  delay(10);
  dmd.fillScreen(0);
  dmd.selectFont(&Arimo_Bold_14_font);
  String s = "2.0";
  dmd.drawString(x, y, s.c_str(), s.length(), white, 0);
  dmd.swapBuffers(true);
  }
 y = y + 19;
}
}
```

ravigyanvihar, 2024-10-14:
If that's the case may be the problem is the template DMD_RGB_SHIFTREG_ABC<RGB80x40plainS20, COLOR_1BITS> 

I ran all orientation but still problem persist with .

I will retry and let you know .

closed 2024-11-09 by board707