board707/DMD_STM32 #132: Examples are not working

krishnaTORQUE, 2024-10-05:
None of the examples are working.

```
Compilation error: 'mux_list' was not declared in this scope
```

board707, 2024-10-05:
Hi @krishnaTORQUE 
I just tested a some examples and all of them were compiled fine.

Could you provide a more info about the problem? What the board do you use? What the version of the DMD_STM32 library?
Please show a full output for compiling errors.

krishnaTORQUE, 2024-10-06:
This one: https://github.com/board707/DMD_STM32/blob/dev-V2/examples/STM32F4_RP2040/dmd_rgb/dmd_rgb.ino

I just copy & paste the entire code & tried to compile & got the error.

board707, 2024-10-06:
What the board do you use?

krishnaTORQUE, 2024-10-06:
STM23F411CEU

board707, 2024-10-06:
Please show the compiler message in full

krishnaTORQUE, 2024-10-07:
Github is preventing me to give you full debug log due to very large amounts of characters.

board707, 2024-10-07:
You could pack it as ZIP and attach to the message

krishnaTORQUE, 2024-10-07:
[Uploading debug-log.txt…]()

board707, 2024-10-07:
You using a wrong STM32 Arduino package. The library is compatible with Roger Clark's [Arduino_STM32 package](https://github.com/rogerclarkmelbourne/Arduino_STM32) only.
Please see the library [Quick Start guide](https://github.com/board707/DMD_STM32/wiki/quick_start) for more details.

closed 2024-10-13 by board707