board707/DMD_STM32 #133: Add a new panel

Danilo277, 2024-10-16:
## Panel description 
#### Dimensions (in pixels) : 32x16
#### Scan factor: 1/8
#### Chips on rear side:  DP245D, DP74HC138B, SM16208SC

## Panel photos
[image]

## Sample videos
When uploaded "dmd_rgb_pattern_test" and only change "RGB32x16plainS4" to "RGB32x16plainS8".

https://github.com/user-attachments/assets/6a857c3a-11ff-4e9b-9577-05477063ccf2

## Specifications

I've tried a lot of combinations of settings but I'm literally nowhere near a solution :).

  - Library version: 1.1.5
  - Platform: STM32
  - MCU board: bluepill STM32F1
  - Connections: used pins - default

board707, 2024-10-16:
Hi
In general, 32x16 panels with 8scans doesn't require any additional setting to work. Please show your code in full.

Danilo277, 2024-10-20:
Thank you for the response!

The problem was a short circuit on the PCB. It was a manufacturing defect... everything works great now.
Best regards.

closed 2024-10-20 by Danilo277