board707/DMD_STM32 #134: At starting ON condition display get noise

maxmurugan, 2024-10-16:
hi sir
how are you?

i am using pico same harware working fine red color and full color 

P4 RGB qiangli module DP5135D+RUL5158C   -- working fine no issue

**change new patch ic DP5125D+DP32020A  -- at initial time get noise after few minute automatically remove noise
how to remove noise ?**

both are working same setting ---      DMD_RGB_SHIFTREG_ABC <RGB80x40_S10_LNikon, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

please help
[image]

maxmurugan, 2024-10-16:
DP5125+DP32020 only issue happened check video
hardware side fine

https://github.com/user-attachments/assets/4e572650-a273-4510-993b-39600933e54a

board707, 2024-10-16:
Hi
Full initialization of all data in the code happens in less than a second... Also, your other matrix works... so I don't think the problem is in the code.
It looks more like some electronic transient processes. For example, charging a large capacitor in the power supply. Or a bad contact that heats up and turns on the circuit.

maxmurugan, 2024-10-24:
hi sir

Issue happened ( DP5125+DP32020 )  this patch only
I try to change another 2 to 3 model working fine .

thanks for your support

closed 2024-11-09 by board707