board707/DMD_STM32 #135: Как запустить P3-320x160-S13

aidarishkulov, 2024-10-19:
## Panel description 
#### Dimensions (in pixels) : 104x52
#### Scan factor: 1/13
#### Chips on rear side:  SM5166, DP5125

  - Library version: 1.01
  - Platform: RP2040 Zero
  - MCU board: blackpil STM32F411
  - Connections: 
  - #define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 2
#define ENABLE_DUAL_BUFFER 0
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
#define DMD_PIN_OE 15
#define DMD_PIN_SCLK_LAT 12
#define DMD_PIN_CLK 11
#define DMD_PIN_R1 0
#define DMD_PIN_G1 1
#define DMD_PIN_B1 2
#define DMD_PIN_R2 3
#define DMD_PIN_G2 4
#define DMD_PIN_B2 5

Для улицы приобрел на [Р10.ru](http://out.arduino.ru/?redirect=http%3A%2F%2Fxn--10-tjc.ru%2F&baseU=https%3A%2F%2Fforum.arduino.ru%2Ft%2Fdmd-stm32-biblioteka-dmd-matricz-dlya-stm32-i-rp2040%2F8573%2F621) панели Р3-320х160-13S с чипами DP5125 и SM5166. Запускал с R
GB104x52_S13_Craftish 4,104,52,13,59. Но, оказывается, там чипы не стандартные, у меня стандартные, тут читал об этом. Если изменить 59 и перебрать 1, 2, 3 должен заработать?

https://github.com/user-attachments/assets/295ef507-aa5f-4d52-8939-b0e782fbf34d

board707, 2024-10-19:
Здравствуйте
покажите код, с которым записано это видео

aidarishkulov, 2024-10-20:
Здравствуйте!
`#include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

/* =================== *
*     STM32F4 pins     *
*  =================== */
#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be consecutive in ascending order
uint8_t custom_rgbpins[] = { 11, 0,1,2,3,4,5 }; // CLK, R0, G0, B0, R1, G1, B1
#endif

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 Direct mux
// RGB32x16plainS2         - 32x16 1/2 Binary mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Direct mux

//DMD_RGB <RGB32x16plainS4, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB <RGB104x52_S13_Craftish, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}`

aidarishkulov, 2024-10-20:
Второй скетч:
`#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false
/* =================== *
*     STM32F4 pins     *
*  =================== */
#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// контакт OE должен быть одним из PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Контакты для каналов R0, G0, B0, R1, G1, B1 и для часов.
// По умолчанию библиотека использует порядок цветов RGB.
// Если вам нужно это изменить — измените порядок контактов R0, G0, B0, R1, G1, B1.
// Все эти контакты также должны быть выбраны из одного порта!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be consecutive in ascending order
uint8_t custom_rgbpins[] = { 11, 0,1,2,3,4,5 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
[... 160 lines cut ...]

                    }
                    else {
                        if (step[i] < 0) dmd.drawMarqueeX(m, 0, dmd.height());
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

                // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

         
        }
    }
}
`

aidarishkulov, 2024-10-20:
Второй скетч:
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false
/* =================== *
*     STM32F4 pins     *
*  =================== */
#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// контакт OE должен быть одним из PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Контакты для каналов R0, G0, B0, R1, G1, B1 и для часов.
// По умолчанию библиотека использует порядок цветов RGB.
// Если вам нужно это изменить — измените порядок контактов R0, G0, B0, R1, G1, B1.
// Все эти контакты также должны быть выбраны из одного порта!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be consecutive in ascending order
uint8_t custom_rgbpins[] = { 11, 0,1,2,3,4,5 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
[... 159 lines cut ...]
                           (dmd.height() > 16)? 8 : 0, 1);

                    }
                    else {
                        if (step[i] < 0) dmd.drawMarqueeX(m, 0, dmd.height());
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

                // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

         
        }
    }
}

board707, 2024-10-21:
Добрый день
Похоже что ваша матрица должна быть совместима с паттерном RGB104x52_S13_Craftish . Проблема в том, что этот паттерно только для плат STM32.  В коде для РП2040 используется другой принцип работы с матрицами и паттерн с ним не совместим. 
Я попробую адаптировать паттерн для РП2040, когда появится свободное время. К сожалению, у меня сейчас очень много работы.

aidarishkulov, 2024-10-21:
Спасибо за ответ! Попробую с STM32F4.

aidarishkulov, 2024-11-07:
Получил STMF103 и STMF411. Попробовал запустить этот код:
`/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS> 
// <COLOR_1BITS> mode is not recomended for bitmap

DMD_RGB <RGB104x52_S13_Craftish, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    // uncomment the line below if you using St-Link with STM32F103xx boards
    //enableDebugPorts();
    
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(5);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(8);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}`
https://github.com/user-attachments/assets/8f4387a9-3532-4f6c-be4a-aeb56561963a

aidarishkulov, 2024-11-07:
STM32F103 

https://github.com/user-attachments/assets/11248294-8090-411b-af6d-a48707498860

board707, 2024-11-08:
Hi
Thanks for the videos.
Your code is too fast to view something. Please increase the delay inside a `for` loops in the `loop()` function. You use a `delay(8)` now, change it to delay(30) or even delay(50). 
There is no need to record the process of completely filling the panel, 1-2 minutes is enough .

aidarishkulov, 2024-11-08:
Снял.

https://github.com/user-attachments/assets/238f84f7-6642-4f5d-85a5-d3573734abe3

board707, 2024-11-09:
Что-то тут не так. Вы точно не забыли увеличить задержку?

board707, 2024-11-09:
Нет, оно как-то совсем неправильно работает.
Уточните, это та самая панель, про которую Вы писали в начале?

> Dimensions (in pixels) : 104x52
> Scan factor: 1/13
> Chips on rear side: SM5166, DP5125

aidarishkulov, 2024-11-10:
Да, эта панель.

aidarishkulov, 2024-11-10:
Добавил в скетч после библиотеки
#define RGB104x52_S13_Craftish	 4,104,52,13,3

Uploading VID-20241111-WA0012.mp4…

aidarishkulov, 2024-11-10:
Добавил в скетч после библиотеки
#define RGB104x52_S13_Craftish	 4,104,52,13,3

https://github.com/user-attachments/assets/94aafce2-20ad-4a72-8cde-25e7eb9e7c78

aidarishkulov, 2024-11-10:
Если поменять в конце 3 на 2 так получается 

https://github.com/user-attachments/assets/4a86b96e-1bb4-487d-923a-622cc3dba236

board707, 2024-11-11:
Отлично!
Это значительно лучше! Сегодня уже поздно, завтра посмотрю внимательно.

board707, 2024-11-11:
Попробуйте описать паттерн так
#define RGB104x52_S13_Craftish 4,104,52,13,112,8,PATTERN_LNЫ

aidarishkulov, 2024-11-11:
> Попробуйте описать паттерн так #define RGB104x52_S13_Craftish 4,104,52,13,112,8,PATTERN_LNS

https://github.com/user-attachments/assets/00c48d2d-d26d-4514-9d4a-853752b90427

aidarishkulov, 2024-11-11:
#define RGB104x52_S13_Craftish 4,104,52,13,102,4,PATTERN_UPS

https://github.com/user-attachments/assets/5c3f9b0b-19fe-4df3-ad16-818f7d0bf196

board707, 2024-11-11:
Не надо редактировать строчки. которые я Вам пишу.
Это строка неправильная:

#define RGB104x52_S13_Craftish 4,104,52,13,102,4,PATTERN_UPS

Перед пераметром  PATTERN_UPS должна стоять цифра 8, а не 4.

aidarishkulov, 2024-11-11:
Видео выше относится Вашей строки, другое видео с моей строкой удалил

board707, 2024-11-11:
Ясно. спасибо.
Похоже не работает.
Попробуйте заменить в этой строке цифру 112 на 111, 101 и 102. Но если заполнение будет опять по 4 точки, то можете видео не записывать - значит шаблон не подходит.
Наверно я удаленно не смогу решить эту задачу.

aidarishkulov, 2024-11-11:
> Ясно. спасибо. Похоже не работает. Попробуйте заменить в этой строке цифру 112 на 111, 101 и 102. Но если заполнение будет опять по 4 точки, то можете видео не записывать - значит шаблон не подходит. Наверно я удаленно не смогу решить эту задачу.
Проверил всё по 4 выводится.
Нельзя панель Вам отправить для теста? Вы в каком городе?

board707, 2024-11-11:
Я в Москве

aidarishkulov, 2024-11-11:
Можете скинуть адрес с куда отправить панель? proekt759@gmail.com

aidarishkulov, 2024-11-22:
Спасибо автору Дмитрию за адаптирование вышеуказанных панелей  библиотеку, все работает хорошо.
Скоро панель появится в библиотеке.

aidarishkulov, 2024-11-22:
Добрый день! Не планируете адаптировать библиотеку в плату STM32 от Espressif ?
Кларка ядро не поддерживается уже. В проекте может давать ошибку
>Message ID:  <board707/DMD_STM32/issues/135/2468509341 @ github . com> 
 
 
--
Айдар Ишкулов
Отправлено из Почты  Mail

board707, 2024-11-22:
> Не планируете адаптировать библиотеку в плату STM32 от Espressif ? Кларка ядро не поддерживается уже.

В принципе когда-то стояло в планах, но сейчас отодвинул на дальнюю перспективу. спрос на СТМ32 маленький, в основном все новые проекты делаются на ЕСП32 и РП2040.
А в ядро Кларка я сам иногда дописываю то, что требуется :)

closed 2024-12-12 by board707