board707/DMD_STM32 #136: New panel pattern test - strange display result with different color

xuniuer, 2024-10-24:
Hi, 

I am trying to find a proper panel template using `dmd_rgb_pattern_test.ino` for my outdoor P8 32x16_4S panel. 

I have tried all predefined `RGB32x16 ... S4` pattern, no luck. So I try to use a new pattern
`#define RGB32x16_s4_xuniuer99     	2,32,16,4,112,8,PATTERN_UND` 
then to run the pattern test program, I got a good result. The led is ON with BLUE color row by row.

Howerver, when I change foreground color to {255, 0, 0} or {0, 255, 0}, I found:
when set `fg = dmd.Color888(255, 0, 0)`, the upper half is RED but the lower half is `GREEN`;
when set `fg = dmd.Color888(0, 255, 0)`, the upper half is GREEN but the lower half is `RED`.
The above the sequence of LED ON is correct like when the `fg` is set to BLUE.   

Plus: 
1) panel IC is `FM6124B`
2) Driver broad is `RPI2040 PICO`.

xuniuer, 2024-10-24:
Attached is a short video.

https://github.com/user-attachments/assets/0c1189eb-8aba-4a1a-8486-371e6cfce26a

board707, 2024-10-24:
Hi
I guess nothing in the code can explain this. It's definitely a hardware issue.
It looks like the R2 and G2 pins are swapped - either when connected or on the matrix itself.

Try to swap the corresponding wires and see the result.

xuniuer, 2024-10-24:
So smart you are! 

After checking, your suggestion is absolutely right! 
Thanks.

xuniuer, 2024-10-24:
Thanks again, I will close the issue.

closed 2024-10-24 by xuniuer