board707/DMD_STM32 #138: Add a new panel - 80x40, 10s, outdoor, ic - FM6124

xuniuer, 2024-11-01:
Hi @board707 

I have a new outdoor panel with the resolution of 80 x 40 and 1/10 scan. The panel IC is FM6124.
[image]

I use the pattern test program (based on RP2040 pico) with the following parameters to init dmd.
`
#define RGB80x40_s10_pt1 4,80,40,10,1
typedef DMD_RGB<RGB80x40_s10_pt1, COLOR_4BITS> DMD_RGB_80_40_10s_4bits;
DMD_RGB_80_40_10s_4bits dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(10);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
`

I've got the following the result:

https://github.com/user-attachments/assets/c6865009-3ca4-43fc-a724-399cf5ffbbb7

I dont' see the leds are not ON row by row, then how to config the parameter `4,80,40,10,1` to support the panel, please advise.

Thanks

board707, 2024-11-01:
Hi
It seems that your panel has one of the known pattern.
Try use this:

       4,80,40,10,102,1,PATTERN_UPS

xuniuer, 2024-11-01:
I have a try using your advise, the config makes the lower rows of upper half panel  ON row by row firstly. 
I change to use ```4,80,40,10,102,1,PATTERN_LPS```, it works. 

Thank you.

board707, 2024-11-01:
Sorry, I was unable to distinguish between LPS and UPS patterns in the video.

xuniuer, 2024-11-01:
Yes, it's difficult to tell the difference from video. Take it easy.

Thanks to the library and its great arch, it's very quick and easy to support a new panel.  Thank you very much.

closed 2024-11-01 by board707