board707/DMD_STM32 #139: RP2040 PICO: setBrightness() doesn't work

xuniuer, 2024-11-14:
Hi @board707 ,

I test the function `setBrightness()` on RP2040 PICO board, it seems the function doesn't work. 
That is, the brightness is always same whaterver to setBrightness(0) or to setBrightness(255).

I try to find only for STM32 case `this->brightness` is used. For RP2040, it's forgotten.

I don't if the case is an issue.

Thanks.

board707, 2024-11-14:
Hi
Which panel are you experiencing this issue with? Not all drivers support brightness control.

xuniuer, 2024-11-14:
I use P4 outdoor，80x40，10S，FM6124.

Thanks.

---- Replied Message ----
| From | ***@***.***> |
| Date | 11/14/2024 17:12 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [board707/DMD_STM32] RP2040 PICO: setBrightness() doesn't work (Issue #139) |

Hi
Which panel are you experiencing this issue with? Not all drivers support brightness control.

xuniuer, 2024-11-14:
Sorry, I cannot do it now.

Tomorrow morning at Beijing time, I will upload my test code.

Thanks.

---- Replied Message ----
| From | ***@***.***> |
| Date | 11/14/2024 18:01 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [board707/DMD_STM32] RP2040 PICO: setBrightness() doesn't work (Issue #139) |

Could you show the code that you used for test?

board707, 2024-11-16:
Hi
I tested your code on two different panels (one with simple pattern and another with complex one, like yours).
The code works fine for me.

Keep in mind that brightness changes logarithmically, so changes at the beginning of the scale are more noticeable. In your code, for example, from brightness 30 to 128 the changes are almost imperceptible, while from 1 to 20 they are very visible.

To make the changes easier to notice, I suggest changing the brightness parameter also according to the logarithmic law:

```
if (brightness > 128) brightness = 1; 
    else brightness += 1 +  brightness/10 ;
```

xuniuer, 2024-11-16:
Thanks for your efforts and suggestion.
I will have a try in the coming Monday.

BTW, I use platformio to build my project in which the DMD lib version 1.1.5 is used. And, your test is also based on the version, right?

---- Replied Message ----
| From | ***@***.***> |
| Date | 11/16/2024 18:12 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [board707/DMD_STM32] RP2040 PICO: setBrightness() doesn't work (Issue #139) |

Hi
I tested your code on two different panels (one with simple pattern and another with complex one, like yours).
The code works fine for me.

Keep in mind that brightness changes logarithmically, so changes at the beginning of the scale are more noticeable. In your code, for example, from brightness 30 to 128 the changes are almost imperceptible, while from 1 to 20 they are very visible.

To make the changes easier to notice, I suggest changing the brightness parameter also according to the logarithmic law:

if (brightness > 128) brightness = 1; 
    else brightness += 1 +  brightness/10 ;

xuniuer, 2024-11-18:
Hi @board707 ,

Now I update the library to current version - dev-V2.
And I try to use the logarithmic law to set brightness but no luck. I don't know why it cannot work for my case. Maybe the problem is about my panel.

Now I try to use color value to reflect the brightness change. Brightness value is 0 ~ 100 on behalf of brightness percentage. Then to use R,G,B to multiple the percentage to get a new R,G,B.  The result is not accurate but the brightness changes indeed. 

Thanks.

board707, 2024-11-18:
Please explain in more detail what happens when the brightness changes? For example, if the brightness changes not gradually, as in your code, but immediately from brightness 1 to brightness 100? 

> Maybe the problem is about my panel.

Your panel should not affect the operation of this algorithm, because it does not depend on the type of panel driver.

xuniuer, 2024-11-19:
> Did you test the brightness control with the same panel on STM32 based board?

Not yet. I will have a try some day.

---

I just to define a new function to change color values by brightness, like this:

```C++

typedef struct {
  uint8_t r, g, b;
} RGB;

uint16_t color_by_brightness(const RGB color, uint8_t brightness) {
    int r = (int)(1. * color.r * brightness / 100);
    int g = (int)(1. * color.g * brightness / 100);
    int b = (int)(1. * color.b * brightness / 100);

    return dmd.Color888(
        static_cast<uint8_t>(r), 
        static_cast<uint8_t>(g),
        static_cast<uint8_t>(b)
    );
}
```
I think the `brightness`(range: 0 ~ 100) is the percentage of color value. 
```
   100% -> 255
   50% -> 127 
   ...
```
The final color are not accurate but the difference, for me, could be ignored.

board707, 2024-11-20:
Hi
Couldn't you figure out anything about the brightness issues?
Maybe try another rp2040 board, they glitch quite often...

xuniuer, 2024-12-17:
Hi @board707 

I have confirmed the above `issue` (setBrightness() does'nt work) is false. It works!

Reviewing my wiring, it seems that wiring with many dupont lines is not good approach. 
So, I make a simple PCB to fix the wiring. Today I tested it, the problem disappears.  

[image]

Thanks.

board707, 2024-12-17:
Thank you for your feedback.

closed 2024-12-17 by board707