board707/DMD_STM32 #140: failed to complie for stm32

ravigyanvihar, 2024-11-22:
Sir,
i am trying to run fm6353 driver .
i am using  https://github.com/rogerclarkmelbourne/Arduino_STM32 for stm32 and stm32f401cc board.
but when i compile following error pops.

code 
#include "DMD_RGB.h"
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

error 
Compiling sketch...
"C:\\Users\\Administrator\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -g -Os -Wall -MMD -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DERROR_LED_PIN=PC13 -mcpu=cortex-m4 -DARDUINO=10607 -DVARIANT_blackpill_f401 -DARDUINO_ARCH_STM32F4 -DLED_BUILTIN=PC13 -DCRYSTAL_FREQ=25 -DNO_CCMRAM -DMCU_STM32F401CC -DSERIAL_USB -DUSER_ADDR_ROM=0x08000000 -mthumb -D__STM32F4__ -DSTM32F4 "-IC:\\Users\\Administrator\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F4\\system/libmaple" "-IC:\\Users\\Administrator\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F4\\cores\\maple/libmaple/usbF4" "-IC:\\Users\\Administrator\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F4\\cores\\maple" "-IC:\\Users\\Administrator\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F4\\variants\\blackpill_f401" "-Ic:\\Users\\Administrator\\Documents\\Arduino\\libraries\\DMD_STM32" "-Ic:\\Users\\Administrator\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "-IC:\\Users\\Administrator\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F4\\libraries\\SPI\\src" "C:\\Users\\Administrator\\AppData\\Local\\Temp\\arduino\\sketches\\F2A147237C33DAAECAEB78A48EF32970\\sketch\\sketch_nov22a.ino.cpp" -o "C:\\Users\\Administrator\\AppData\\Local\\Temp\\arduino\\sketches\\F2A147237C33DAAECAEB78A48EF32970\\sketch\\sketch_nov22a.ino.cpp.o"
In file included from c:\Users\Administrator\Documents\Arduino\libraries\DMD_STM32/DMD_RGB.h:24:0,
                 from C:\Users\Administrator\AppData\Local\Temp\.arduinoIDE-unsaved20241022-23364-9qr169.08lrs\sketch_nov22a\sketch_nov22a.ino:1:
c:\Users\Administrator\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:368:2: error: 'PortType' does not name a type
  PortType clk_clrmask = 0;
  ^~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:369:2: error: 'PortType' does not name a type
  PortType clkmask = 0;
  ^~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:379:2: error: 'OE_PWM_Polarity' does not name a type
  OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE;
  ^~~~~~~~~~~~~~~

Using library DMD_STM32 at version 1.1.2 in folder: C:\Users\Administrator\Documents\Arduino\libraries\DMD_STM32 
Using library Adafruit GFX Library at version 1.7.0 in folder: C:\Users\Administrator\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library SPI at version 1.0 in folder: C:\Users\Administrator\Documents\Arduino\hardware\Arduino_STM32\STM32F4\libraries\SPI 
exit status 1

Compilation error: exit status 1

board707, 2024-11-22:
Hi
I was unable to reproduce the error, your code compiled fine for me.

I can only guess, why you are getting errors.
 For example, I see that you did not save the sketch before compiling:
```
C:\Users\Administrator\AppData\Local\Temp.arduinoIDE-unsaved20241022-23364-9qr169.08lrs\sketch_nov22a\sketch_nov22a.ino
```
This situation is known to sometimes lead to strange glitches.

Try saving the code as a new Arduino sketch, restarting the IDE and selecting the STM32 board in the settings again

ravigyanvihar, 2024-11-25:
no sir , i am still getting same error
can you tell me which version you are using for stm32 board
i cloned it from
https://github.com/rogerclarkmelbourne/Arduino_STM32

but may be any recent update is causing the issue .

. also have you fingered out commands for "fm6363" . 

actually since last few month gen2 panels are becoming extremely hard to find . 
i have beeing trying to put gen3 into work .
i using your work as base .

board707, 2024-11-26:
> i am still getting same error

Do you try to compile the same empty code?
```
#include "DMD_RGB.h"
void setup() {}

void loop() {}
```
Please update DMD_STM32 library to the latest version (1.1.5/1.1.6).

You could clone STM32 package from my github
https://github.com/board707/Arduino_STM32

For installation instructions, see
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

board707, 2024-12-04:
Hi @ravigyanvihar 
Were you able to solve the problem?

ravigyanvihar, 2024-12-05:
No sir ,
Still same issue

board707, 2024-12-17:
Where did you edited the macro?

ravigyanvihar, 2024-12-17:
In Library 
DMD_STM32a.h 
Line number 63 

DMD_Monochrom_Parallel.h 
Line number 30

board707, 2024-12-18:
Line 63 of DMD_STM32a.h doesn't contain this macro.
It seems to me that the library you compiled is not mine.

closed 2025-02-05 by board707