board707/DMD_STM32 #142: 128x128 Panel, FM6363, ICND2018, HUB75

Jan1503, 2025-01-22:
Hi there,
would you mind taking a look at this discussion:
[https://github.com/LAutour/ESP32-HUB75-MatrixPanel-I2S-DMA-icn2053/discussions/6](https://github.com/LAutour/ESP32-HUB75-MatrixPanel-I2S-DMA-icn2053/discussions/6)

This is an old thread of mine discussing the correct initializing of these panels.

I wonder if your library could drive those nice panels (2 or more at best) using an RPi2040 or 2350.

Have two of them laying around in some dark place in the cellar and would love to spend some time on it again :)

Kind regards,
Jan

board707, 2025-01-22:
Hi
In general I am familiar with this discussion and even had a conversation with the author of the code at that time.
I think your panel (at least a single) can work with my library, but keep in mind that support for the FM6363 driver is only for boards on the STM32 platform. Also, the library uses only 4-bit color.

Honesty speaking, these drivers are not for every-day use yet. In all github projects that I know, support for PWM drivers works only as option and requires configuration in each individual case. Supporting them requires a lot of work, and there is almost no demand for it. In the year since I announced PWM driver support in my project, as far as I know, only one person has used it.
 I'm rather considering developing a code for drivers 6353 and 6363 as a programming challenge :) .

Now I am worked on figuring out a protocol of DP3264 PWM driver.

Jan1503, 2025-01-23:
Hi and thanks for your quick reply.

I will try the code, just had to buy an STM board first. Thought it would work with a RPi. Which one do you recommend?

Speaking of the DP3264 driver, I just talked to a chinese supplier (Coreman) about [this](https://a.aliexpress.com/_Exj5BsM) panel, which uses a DP3265S PWM chip with a super-nice resolution of 256x128 in a P1.25 pitch package.

Would buy one if you think, you could send me some testing code :)

Jan1503, 2025-01-23:
I'm thinking more and more to move away from this DIY approach but sadly, there seems to be no system on the market today capable of using your own "sender", as each manufacturer uses a different protocol.
I know there was a receiver card from Colorlight once, model 5A-75B, but I doubt it's able to handle modern displays as it is very old.
It looks like the protocol has been reversed engineered so you jsut write your own code to send data to it.
I mainly use .net running Henners library on a RPi4b with good results.
Will continue to watch your repo for updates...these flexible display from above really catches my attention ;)

board707, 2025-01-23:
> I'm thinking more and more to move away from this DIY approach

I think you are right.
Most DIY controllers has a limited RAM and processing power, not enough to work with big screens. Most DIY projects can manage only one or two panels, if we want to use a color depth and FPS, similar to commercial screens. The best result should be expected applying a FPGA boards, as it used in commercial receiving cards, but dealing with it is too much for DIY .

The application area for DIY LED screens is a various text and information boards, like a clocks, meteorology info, sports score displays and so on. 
All of this does not require new technologies of fast PWM led drivers...

Jan1503, 2025-01-23:
> The typical application area for DIY LED screens is a various text and information boards, like a clocks, meteorology info, sports score displays and so on.

Yes, I know, but for me the real deal is building a nice high-resolution low-pitch display like the one I built in 2023:
https://github.com/user-attachments/assets/7ff5522a-c5b2-4e17-8742-a319f07f93bb

This panel is using P2 panels with an overall 384x192 resolution and is fed by a .net client application using a highly optimized udp-transmission protocol.

I would love to rebuild this panel with say P1.25mm pitch panels and even higher resolution.
But I think this isn't possible with a RPi as long as SPWM-chips are not fully implemented. In the early days with obly R/G displays I developed these drivers by myself, which was easy, but I then moved to other languages and now I'm not smart enough anymore for this stuff.
Feel free to close this "issue" and have a nice day.

closed 2025-01-23 by board707