board707/DMD_STM32 #143: Issue with New P4 panel, please add it

Cyberster, 2025-01-23:
## Panel description 
It is a outdoor P4 panel manufactured by Quangli

#### Dimensions (in pixels) : 80x40

#### Scan factor: 1/10

#### Chips on rear side:  DP5125D, DP32020A, DP245D

## Panel photos
[image]
[image]
[image]
[image]
[image]

## Sample videos
[image]

## Specifications

  - Library version: v1.1.5
  - Platform: STM32
  - MCU board: blackpil STM32F411CEU6
  - Connections: used pins
  -- Default
```
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7 // lat
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1
```

Cyberster, 2025-01-23:
I've tried all patterns vailable in DMD_PANEL_TEMPLATES_H

board707, 2025-01-23:
Your panel has a shift register line driver, so you need to use a `DMD_RGB_SHIFTREG_ABC` class instead of `DMD_RGB`.

Set the template
```
#define RGB80x40plainS10_test  3,80,40,10,1
```
and run a `dmd_rgb_pattern_test` example and show the video.

Cyberster, 2025-01-25:
I have tried the RGB80x40plainS10_test with DMD_RGB_SHIFTREG_ABC class and below is the result.

[image]

```
#define RGB80x40plainS10_test  3,80,40,10,1

// DMD_RGB <RGB80x40plainS10_test, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB_SHIFTREG_ABC <RGB80x40plainS10_test, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

board707, 2025-01-25:
Hi
Thanks for your video... but...
The picture in your video is random noise, which indicates bad connections.  It has nothing to do with the shift register or the pixel pattern, sorry.

> I assumed DI as E pin and connecte it to PB8.

Your panel do not has a E pin, because it has only 10 scan lines, for addressing which 4 pins (A-D) are enough