board707/DMD_STM32 #145: Add a new panel

funnymind, 2025-02-15:
## Panel description 
Quangli P3
#### Dimensions (in pixels) : 104x52
#### Scan factor: 1/32
#### Chips on rear side:  dp5135, rul5158c

## Panel photos

[image]
[image]
[image]
[image]

## Sample videos
#define RGB104x52_S13_t 4,104,52,13,2
DMD_RGB_SHIFTREG_ABC  <RGB104x52_S13_t , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
https://github.com/user-attachments/assets/0edd147c-8374-4a41-bf0a-91df5f96bfea

DMD_RGB_SHIFTREG_ABC  <RGB104x52_S13_Craftish, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
https://github.com/user-attachments/assets/297c64eb-b1c2-409e-bb6a-97e032809687

## Specifications
   
  - Library version: 1.01
  - Platform: STM32
  - MCU board: blackpil STM32F411
  - Connections: used pins

Hi, can you help please with this panel?

board707, 2025-02-16:
Hi
Could you please show the video for pattern 
RGB104x52_S13_t 4,104,52,13,1
?

funnymind, 2025-02-16:
Here, 
#define RGB104x52_S13_t 4,104,52,13,1
DMD_RGB_SHIFTREG_ABC  <RGB104x52_S13_t , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

https://github.com/user-attachments/assets/d2218257-3c05-4952-b808-8d8d8a3a5842

board707, 2025-02-16:
Thank you.
I will try to figure it out in few days.

funnymind, 2025-02-17:
Cool, thanks a lot.

board707, 2025-02-19:
Hi
Please add the code below just after line 
`#include "DMD_RGB.h"`
in your code.

```
#define RGB104x52_S13_funnymind   4,104,52,13,63

template<int COL_DEPTH>
class DMD_RGB<RGB104x52_S13_funnymind, COL_DEPTH> : public
	DMD_RGB_BASE2<COL_DEPTH>
	{
	public:
		DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t*
			pinlist,
			byte panelsWide, byte panelsHigh, bool d_buf = false) :
			DMD_RGB_BASE2<COL_DEPTH>(4, mux_list, _pin_nOE, _pin_SCLK, pinlist,
				panelsWide, panelsHigh, d_buf, COL_DEPTH, 13, 104, 52)
			{
			this->fast_Hbyte = false;
			this->use_shift = false;
			}
		// Fast text shift is disabled for complex patterns, so we don't 
		// need the method
			void disableFastTextShift(bool shift) override {}

	protected:
		uint16_t get_base_addr(int16_t& x, int16_t& y) override {
			this->transform_XY(x, y);
			uint8_t pol_y = y % this->pol_displ;
			x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
                        
                        static const uint8_t px_base = 4; 
			static const uint8_t C_tbl[52] = 
			      { 0,3,28,29,30, 31,32,33,34,35, 36,37,38,39,40, 41,42,43,44,45, 46,47,48,49,50, 51,
                                1,2,4,5,6, 7,8,9,10,11, 12,13,14,15,16, 17,18,19,20,21, 22,23,24,25,26,27 };
			static const uint8_t Pan_Okt_cnt = this->DMD_PIXELS_ACROSS / px_base;
			
                        uint8_t Oktet_m = (pol_y / this->nRows) * Pan_Okt_cnt + (x / px_base) % Pan_Okt_cnt;

			uint16_t base_addr = (pol_y % this->nRows) * this->x_len  +
                       (x /this->DMD_PIXELS_ACROSS) * this->DMD_PIXELS_ACROSS * this->multiplex +
				C_tbl[Oktet_m] * px_base;
			base_addr += x % px_base;

			return base_addr;
			}

	};
```

funnymind, 2025-02-19:
Looks like a miracle, it works from the first try), thank you very much for your help.

[image]

board707, 2025-02-19:
I wish you a new interesting projects!

closed 2025-02-19 by board707

board707, 2025-03-11:
Addede in the library in 9353609