board707/DMD_STM32 #149: Add a new panel  QiangLi Q4S20B20V-80X40

sapfir63, 2025-03-11:
## Panel description 
#### Dimensions (in pixels) : 80x40
#### Scan factor: 1/8
#### Chips on rear side:   DP5135, DP245D  DP32020A

## Panel photos

## Sample videos

## Specifications

  - Library version:  v1.1.5
  - Platform: STM32
  - MCU board: STM32F407 DISCO
  - Connections: 
  - #define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

board707, 2025-03-11:
Hi
Did you try to run library code with your panel? Do you see anything on the panel with this setup?

sapfir63, 2025-03-12:
Добрый день 
да я пробовал запустить  панель,
но она на запустилась, похоже в коде есть ошибка, 
я нашел https://arduino.ru/forum/programmirovanie/stm32-dma-memory-gpio-ne-rabotaet-nulevoi-pin твою переписку по запуску кода, по нему пины rgd работают, но кода запускаю код v1.1.5 сигналов на  {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1 вообще никаких нет

Good afternoon 
yes, I tried to launch the panel,
but it didn't start, it looks like there is an error in the code., 
I found https://arduino.ru/forum/programmirovanie/stm32-dma-memory-gpio-ne-rabotaet-nulevoi-pin your correspondence on running the code, rgd pins are working on it, but I'm running code v1.1.5 there are no signals at all for {PA6, PA0,PA1,PA2,PA3,PA4,PA5}; // CLK, R0, G0, B0, R1, G1, B1

board707, 2025-03-12:
Доброе утро
Покажите код, который вы запускали, полностью.

Если я правильно понял, на каком-то старом коде панель работает? Напишите подробнее, что это за код, версия библиотеки...и приведите скетч тоже полностью. 
Спасибо.

sapfir63, 2025-03-12:
скетч стандартный examples/STM32F4_RP2040/dmd_rgb/dmd_rgb.ino
библиотека установлена через ардуино DMD_STM32 1.1.5

board707, 2025-03-12:
@sapfir63 
Простите за настойчивость, но мне нужно видеть код, который вы запускали. Даже если он максимально близок к примеру.

И поясните по матрице 64*32 - вы ее запускали на этой же версии библиотеки 1.1.5 и с ней пример работал? Плата STM32f407 в обоих случаях была та же?

sapfir63, 2025-03-12:
Да любые вопросы задавайте 
1. панель рабочая запускалась на другой плате под другим кодом без проблем
 я сейчас за компом могу любые эксперименты сделать 
код запуска
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32 and RP2040 boards
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false
/* =================== *
*     STM32F4 pins     *
*  =================== */
#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
[... 185 lines cut ...]

                    }
                    else {
                        if (step[i] < 0) dmd.drawMarqueeX(m, 0, dmd.height());
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

                // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

         
        }
#endif
    }
}

board707, 2025-03-12:
> любые вопросы задавайте

Спасибо

Тогда, чтобы нам не ходить кругами, постарайтесь отвечать не "в общем", а на конкретные вопросы.

1. Тот код, который вы показали в предыдущем сообщении - вы использовали для обсуждаемой панели 80*40 ?  Почему спрашиваю - в нем стоит инициализация матрицы 64х32

> панель рабочая запускалась на другой плате под другим кодом без проблем

2. Если не сложно, укажите под каким кодом и на какой плате.
3. Вы не ответили в прошлом сообщениее про ваш успешный запуск панели 64х32. Какой там был слфт, какая версия, какая плата использовалась.
4. Если можно, вставьте фото задней стороны обсуждаемой панели 80х40, достаточно крупное, чтоб читались надвписи.

board707, 2025-03-12:
Если Вам удобнее, можем перейти в почту.

sapfir63, 2025-03-12:
да удобнее

board707, 2025-03-12:
Ответил

board707, 2025-03-17:
Successfully run as 80x40 1/20 with shift register line driver:
```
#define RGB80x40_S10_test        3,80,40,20,1
DMD_RGB_SHIFTREG_ABC <RGB80x40_S10_test, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

closed 2025-03-17 by board707