board707/DMD_STM32 #156: STM32duino support

JeanHenriqueAlves, 2025-04-09:
Are there any plans to add support to the [official SMT32 core](https://github.com/stm32duino/Arduino_Core_STM32) (STM32duino) by STMicroelectronics instead of RogerClark's already deprecated STM32 core? I have some projects which I want to use DMD, and being based on the official core makes them incompatible with this library.

board707, 2025-04-09:
Hello, I am indeed currently porting my library to the STM core. So it will be compatible in the future. 

But it should be noted that despite the fact that RogerClark's package, as you write, is deprecated - it is actually much better written than STM's one. Code compatible with Clark's package is smaller in size and works much faster. STM's package has no advantages over Clark's package, except that it supports a larger set of boards.

JeanHenriqueAlves, 2025-04-10:
Hello, thanks for the fast reply. Sorry, I had no idea that RogerClark's package was actually better. What I actually wanted to do was to fiddle around a bit and see if your library could run on the STM's I had lying around, 32F030C8T6 and 32F100C8T6B. I have these SMD P10 panels and just couldn't make them work with ESP8266 or ESP32, then I realized I also had a fake Maple Mini with 32F103CBT6 and your library came in handy, so thank you so much for saving my hobby projects!

board707, 2025-04-10:
> if your library could run on ...  32F030C8T6 and 32F100C8T6B.

On the first one most likely not, but on the second one you can try, I think it will work

kocinski15, 2026-10-18:
I have run Claude session over the forked repository to aim at Stm32duino compatibility. 
Dropped AdafruitGFX support to focus on the rest.
Left only SPWM chips in the lib, added 565 color with gamma correction to fully utilize SPWM chip without overloading RAM capacity of STMs.
Result as of this moment is compiling code, however it is yet to be tested on the hardware. You can have a look if you like.
https://github.com/kocinski15/PWM_STM32/tree/stm32duino-spwm