board707/DMD_STM32 #159: P2-1515-128X64-32S-S2 - FM6373

sityware, 2025-04-30:
## P2-1515-128X64-32S-S2
#### 128x64
#### 1/32
#### Chips on rear side:  FM6373 , 74HC04D, DP32019B

## Panel photos

[image]

[image]

board707, 2025-04-30:
Hi
Did you try to run your panel with library settings for FM6363 driver?

sityware, 2025-05-02:
I've just tried this without success.

on my project I changed
#include "DMD_RGB_FM6353.h"
and
DMD_RGB_FM6353 <RGB128x64plainS32,

To confirm I had it working with the same project on a 64x32 panel

#include "DMD_RGB.h"
DMD_RGB_FM6353 <RGB64x32plainS16

board707, 2025-05-03:
> I've just tried this without success.

It seems that the driver FM6373 has a different config procedure than 6363 one. Unfortunately, in that case  I won't be able to help without having a datasheet and the FM6373 driver panel on my hands.

closed 2025-05-05 by board707