board707/DMD_STM32 #160: Add a new panel

codev123, 2025-05-01:
## Panel description 
#### Dimensions (in pixels) : 32x16
#### Scan factor: 1/8
#### Chips on rear side:   DP5125D / RUC72258EH4

## Panel photos

[image]

## Sample videos

https://github.com/user-attachments/assets/682b8284-abd6-4221-827e-a06b5fc48382

## Specifications

  - Library version: 1.01
  - Platform: STM32
  - MCU board: stm32f103c8
  - Connections: 
#define DMD_PIN_A PB1
#define DMD_PIN_B PB2
#define DMD_PIN_C PB3
#define DMD_PIN_D PB4
#define DMD_PIN_E PB5
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D, DMD_PIN_E };
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB6
// Pins for R0, G0, B0, R1, G1, B1 channels and for clock
uint8_t custom_rgbpins[] = { PB9, PB10, PB11, PB12, PB13, PB14, PB15 };  // CLK, R0, G0, B0, R1, G1, B1

DMD_RGB<RGB32x16plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

please sir add this panel or suggest me which class should i use . thank you sir

board707, 2025-05-01:
Hi
Your panel seems to be a compatible with the library, but perhaps we need some specific settings for it.

Now I am not at home. I will return on the next week and will try to solve the issue.

codev123, 2025-05-02:
Ok sir

board707, 2025-05-04:
Hi
I have seen your video again.
According with it, you have a problem with A B C lines, especially with a B pin. Please check your connections - both with a pins on the board and on the panel.
Please clarify what the STM32 board did you use? The question arise because a most popular stm32f103c8 board - a Bluepill - doesn't have a PB2 pin, which you wrote used as B line.

codev123, 2025-05-04:
I am using my custom board with stm32f103c8/f103b uC. I am using pb2 pin in my board. I also tested same board with monochrome p10 with your code and used pb2 as b line and it's working great.
But with rgb p10 led i am having the issue i mentioned. I am doing all that on my own designed pcb with hub 75, hub 12 connectors, so no issue with connection as well.

If you are sure about pb2 line then i will cut track and solder another pin and then will inform you if issue resolves, but monochrome working with pb2 as b line.

board707, 2025-05-04:
> If you are sure about pb2 line then i will cut track and solder another pin

No, I'm not sure that the only problem is PB2 pin. But it looks like line B is not working.
Since pins D and E are not needed on this panel, could you swap pins B and D, for example?

codev123, 2025-05-04:
Ok sir. I will cut track and solder d pin on b. It's midnight here,so will update you tomorrow about the progress. Thankyou sir for quick response.

board707, 2025-05-04:
Also, may be the problem is that the line B on the panel itself is broken. Do you have another such panel for testing?

codev123, 2025-05-04:
Yes sir, i have few panels with me, i will check the code first with another panel and if issue persist then move to solution you proposed.

codev123, 2025-05-05:
i have cut down a trace in pcb and shoreted e pin of p10 with b pin thus using pb5 pin as b pin, but issue worsens

https://github.com/user-attachments/assets/e027ca2f-067e-4bff-b732-1775e6785cf0

board707, 2025-05-05:
> shoreted E pin of p10 with B pin

Please do not short B and E pins on the panel side, just swap their wires.
The E pin on your panel, most probably, is shortened to GND internally, so the B line will be connected to GND also and won't work at all.

codev123, 2025-05-06:
i have cut down track for e and connected track to pin b, so that it doesn't short, and made change in code, pin b as PB5. still same issue as shown in above video.

while  without cutting any tracks with 
#define DMD_PIN_A PB1
#define DMD_PIN_B PB2
#define DMD_PIN_C PB3
#define DMD_PIN_D PB4
#define DMD_PIN_E PB5
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D, DMD_PIN_E };
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB6
// Pins for R0, G0, B0, R1, G1, B1 channels and for clock
uint8_t custom_rgbpins[] = { PB9, PB10, PB11, PB12, PB13, PB14, PB15 }; // CLK, R0, G0, B0, R1, G1, B1

i get the first video i provided in issue.

codev123, 2025-05-06:
Thankyou sir