board707/DMD_STM32 #161: Row shifting, pixels not fully off

omoding-joshua, 2025-05-13:
Hey. I am using the p10 monochrome display with the Blue pill Stm32f103c8t6 on version 1.1.5 of the Library.
However it seems like the rows of the individual panel seem to be shifted. For example in this picture and video;

[image]

https://github.com/user-attachments/assets/8c6553d3-99d3-4057-8227-3387754948e5

The display should be showing "024", however, it looks like what should have been on first row have been moved to the fourth row, what should have been on row 9, has been shifted to row 12.
Also the pixels that should be off are not completely off. They have a faint glow.
Please help me figure out what I am doing wrong

board707, 2025-05-13:
Hi
Thanks for your feedback.
Could you please show a full code that produced the issue?

omoding-joshua, 2025-05-13:
Okay. Thanks for responding
/*--------------------------------------------------------------------------------------
 Demo code for DMD panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
// choose between Parallel ans SPI wiring
// comment line below for SPI connection
//#define DMD_PARA 
/*--------------------------------------------------------------------------------------
  Includes
--------------------------------------------------------------------------------------*/
#if defined(DMD_PARA)
#include "DMD_Monochrome_Parallel.h"
#else
#include "DMD_MonoChrome_SPI.h"
#endif
//#include "DMD_MonoChrome_SPI.h"

// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#include "st_fonts/UkrRusArial14.h"
//#include "st_fonts/Arial_Rounded_MT_Bold32_048_057.h"
#include "st_fonts/Microsoft_Sans_Serif16_037_057.h"

#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER true

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be Timer3 PWM pin as PB0,PB1
// if connect as SPI, do not use corresponding MiSO pin - PA6 for SPI1 and PB14 for SPI2
#define DMD_PIN_A PA1
#define DMD_PIN_B PA4
#define DMD_PIN_nOE PB1
//#define DMD_PIN_nOE PA6
#define DMD_PIN_SCLK PA15
//#define DMD_PIN_SCLK PB4

// pins for SPI connect
// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

//=== Config for Parallel connect ====
#if defined(DMD_PARA)
//pins for rows at x axis
// example for two rows
// all those pins must be selected from same port!
//uint8_t pins[] = { PA5, PA7, PA6 };  // CLK , row1, row 2
uint8_t pins[] = { PA5, PA7};  // CLK , row1, row 2

//Fire up the DMD object as dmd
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN,ENABLE_DUAL_BUFFER);
#else
[... 154 lines cut ...]

	0x04, 0x02, 0x01, 0x21, 0x21, 0x52, 0x8c,  
	0x02, 0x04, 0x08, 0x08, 0x08, 0x04, 0x03,   // char (051) '3'

	0x80, 0x40, 0x30, 0x08, 0x06, 0xff, 0x00,  
	0x01, 0x01, 0x01, 0x01, 0x01, 0x0f, 0x01,   // char (052) '4'

	0x3f, 0x21, 0x11, 0x11, 0x11, 0x21, 0xc1,  
	0x02, 0x04, 0x08, 0x08, 0x08, 0x04, 0x03,   // char (053) '5'

	0xfc, 0x22, 0x11, 0x11, 0x11, 0x22, 0xc4,  
	0x03, 0x04, 0x08, 0x08, 0x08, 0x04, 0x03,   // char (054) '6'

	0x01, 0x01, 0x01, 0x01, 0xc1, 0x39, 0x07,  
	0x00, 0x00, 0x00, 0x0e, 0x01, 0x00, 0x00,   // char (055) '7'

	0x8c, 0x52, 0x21, 0x21, 0x21, 0x52, 0x8c,  
	0x03, 0x04, 0x08, 0x08, 0x08, 0x04, 0x03,   // char (056) '8'

	0x3c, 0x42, 0x81, 0x81, 0x81, 0x42, 0xfc,  
	0x02, 0x04, 0x08, 0x08, 0x08, 0x04, 0x03   // char (057) '9'

};

#endif

board707, 2025-05-13:
The last parameter of the drawString() method is a text orientation. You call it with an orientation "1", that means a vertical text:
```
dmd.drawString(0, 0, str, strlen(str),1);
```

but use a standard font, which can't be vertical.

omoding-joshua, 2025-05-13:
This is what I get when I use the dmd_monochrome example. All I changed is change the text to '1234' and increased the interval to 300 to slow down the scrolling

[image]

https://github.com/user-attachments/assets/54133a4c-7ed0-49bf-983f-e24242a1252f
I still have the same issue.
Could you also please explain what the parameter 'color' does in the dmd.drawString() function

board707, 2025-05-13:
> I still have the same issue.

Could you please presents exact code, that produced this video? 
I could to test it on my panels at the evening .

What did you use to upload the code - StLink or Serial with bootloader?
Could you show a purchase link or photo of your STM board  ?

omoding-joshua, 2025-05-14:
> Could you please presents exact code, that produced this video?
I could to test it on my panels at the evening .
okay. Here it is.

/*--------------------------------------------------------------------------------------
 Demo code for DMD panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
// choose between Parallel ans SPI wiring
// comment line below for SPI connection
//#define DMD_PARA
/*--------------------------------------------------------------------------------------
  Includes
--------------------------------------------------------------------------------------*/
#if defined(DMD_PARA)
#include "DMD_Monochrome_Parallel.h"
#else
#include "DMD_MonoChrome_SPI.h"
#endif

// Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER true

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be Timer3 PWM pin as PB0,PB1
// if connect as SPI, do not use corresponding MiSO pin - PA6 for SPI1 and PB14 for SPI2
#define DMD_PIN_A PA1
#define DMD_PIN_B PA4
#define DMD_PIN_nOE PB1
//#define DMD_PIN_nOE PA6
#define DMD_PIN_SCLK PA15
//#define DMD_PIN_SCLK PB4

// pins for SPI connect
// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

//=== Config for Parallel connect ====
#if defined(DMD_PARA)
//pins for rows at x axis
// example for two rows
// all those pins must be selected from same port!
uint8_t pins[] = { PA5, PA7, PA6 };  // CLK , row1, row 2

//Fire up the DMD object as dmd
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN,ENABLE_DUAL_BUFFER);
#else

//=== Config for SPI connect ====
SPIClass dmd_spi(1);
[... 138 lines cut ...]

                    }
                    else {
                        if (step[i] < 0) dmd.drawMarqueeX(m, 0, dmd.height());
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

            // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

           

        }
    }
}

omoding-joshua, 2025-05-14:
> What did you use to upload the code - StLink or Serial with bootloader?
Could you show a purchase link or photo of your STM board ?
I am using the ST Link

board707, 2025-05-14:
Thank you.
I will test your code at the evening .

omoding-joshua, 2025-05-14:
okay

board707, 2025-05-14:
What might help in your case to remove distortion.

- Test all your wire connections. Pay special attention to the GND connection between board and the panel. These LED panels use high frequency signals and are very sensitive to bad connections.

- Second, if you are using ST-link to upload firmware, add this call as the first line in the setup:
```
enableDebugPorts();
```

- Third, if you are running a modified example from the library - be sure to save the sketch under a new name before compiling. There is a bug in the Arduino IDE, due to which examples do not work after editing, if you have not saved them.