board707/DMD_STM32 #164: Scrolling issue

maxmurugan, 2025-05-24:
hi sir 

Pico controller 

char Message_1[1024] = "India’s northeast is not a frontier region any more, it is the “frontrunner of growth, Prime Minister Narendra Modi said on Friday (May 23, 2025), pitching for investments in the eight States that constitute the region.Inaugurating the Rising Northeast Summit, the Prime Minister said that the region offers wide range of economic and industrial opportunities. India has new dreams for the northeast, he said, adding that the region would emerge as a gateway for trade with southeast Asia over the next decade";

using 534 char

only scroll 290char  message only

how to solve this issue

void Update_led_Scroll()
{
    if (first_start) {
      dmd.drawMarquee(Message_1, sizeof(Message_1) , dmd.width() - 5, 0, 0);
      first_start = false;
    }
    //##########################
    if ((millis() - prev_step) > interval)
    {
      prev_step = millis();
      if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
        dmd.clearScreen(true);
        dmd.drawMarquee(Message_1, sizeof(Message_1) , dmd.width() - 5, 0, 0);
      }
      dmd.swapBuffers(true);
    }
}

board707, 2025-05-24:
Hi
To scroll a string more than 300 chars wide you need to change the

#define MAX_STRING_LEN 300

in the file DMD_STM32a.h line 85

maxmurugan, 2025-05-24:
wow it is working
thankyou sir