board707/DMD_STM32 #169: Question regarding RP2040 pinout

JeanHenriqueAlves, 2025-06-17:
Hello, I'm going to upgrade my board to RP2040 and I have some questions about the pinout, since I could not get any of my peripherals to work with STM32F4 and Roger's core. From what I read in the wiki, pins must be set in consecutive ascending order, but I want to know if there is any specific pin sequences that must be followed. I'm planing to use the following pinout, so that I don't lose 2 GPIOS and UART1 to DMD pins:

```
/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 10
#define DMD_PIN_B 11
#define DMD_PIN_C 12
#define DMD_PIN_D 13
// put all mux pins at list
uint8_t mux_list[] = {DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D};

// On RP2040 you can use any GPIO for OE signal
#define DMD_PIN_nOE 20
#define DMD_PIN_SCLK 15

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be consecutive in ascending order
uint8_t custom_rgbpins[] = {14, 0, 1, 2, 3, 4, 5}; // CLK, R0, G0, B0, R1, G1, B1
```

Will this work, or I _must_ place mux pins at 6, 7, 8 and 9? My panel has no E pin so it has been omitted.

board707, 2025-06-17:
> Will this work, or I _must_ place mux pins at 6, 7, 8 and 9?

No, the mux_pins and rgb_pins groups are independent.

JeanHenriqueAlves, 2025-06-18:
Hi, thank you, I'm going to test it as soon as possible.