board707/DMD_STM32 #176: Please, help with panel P6 1/8S (BGR)

sharayura, 2025-09-02:
## Panel description 
#### Dimensions (in pixels) : 32x32
#### Scan factor: 1/8
#### Chips on rear side:  SM16206S, SM74HC138D

## Panel photos
 
[image]

3 панели 32Х32 объеденены последовательно, управлялись платой HD-U60-75, в программе HD2018, сейчас необходимо выводить информацию со специфического датчика I2C
выбиралась опция Hardware [C61] P6 1/8S HUB75 (BGR)
На наклейке панелей P6SMD170616-311J/G3000

## Sample videos
Загрузили example dmd_rgb_pattern_test с шаблоном RGB32x32_S8_maxmurugan. На видео может быть не видно, но все линии прорисовываются по два раза. Пожалуйста, подскажите что можно сделать
https://youtube.com/shorts/RIEEOSYw500

## Specifications

  - Platform: STM32
  - MCU board: blackpil STM32F103C

sharayura, 2025-09-02:
очень рад быстрому ответу!
https://youtube.com/shorts/BBt0UbNpUJc?feature=share

sharayura, 2025-09-02:
Благодарю! со вторым шаблоном более последовательно, но все равно два раза прорисовывает 4 строки, а 4 оставляет пустыми

https://github.com/user-attachments/assets/5b628522-6c46-4507-9d93-68dc714e6ce8

board707, 2025-09-02:
Понятно.
Тогда просьба - запишите мне еще раз видео для тестового шаблона

#define RGB32x32_S8_pt1		3,32,32,8,1	// 32x32 1/8 BINARY 

а заодно и для 

#define RGB32x32_S8_pt2		3,32,32,8,2	// 32x32 1/8 BINARY 

только в этот раз подключите только одну панель.
Не забудьте в коде изменить настройки числа панелей.

sharayura, 2025-09-02:
#define RGB32x32_S8_pt1 3,32,32,8,1 // 32x32 1/8 BINARY

https://github.com/user-attachments/assets/2dc77ea5-3687-4211-91e7-ebaa996635e4

#define RGB32x32_S8_pt2 3,32,32,8,2 // 32x32 1/8 BINARY

https://github.com/user-attachments/assets/c7e70f6a-bc71-429d-b49a-424c200bfd4f

board707, 2025-09-02:
Какая у вас плата СТМ32 ? Как Вы прошиваете код - через Сериал порт или через Ст-линк?
Выложите, пожалуйста, свой тестовый скетч полностью (даже если он практически совпадает с примером).

sharayura, 2025-09-02:
прошиваю через ст-линк, Arduino IDE, плата стандартная с контроллером STM32F103C8T6 

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PB0

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA7, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS> 
// <COLOR_1BITS> mode is not recomended for bitmap

#define RGB32x32_S8_pt2 3,32,32,8,2 // 32x32 1/8 BINARY  

DMD_RGB <RGB32x32_S8_pt2, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    // uncomment the line below if you using St-Link with STM32F103xx boards
    //enableDebugPorts();
    
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}

sharayura, 2025-09-02:
вот это да! огромное спасибо, прошу прощения за мою невнимательность!
с шаблоном RGB32x32_S8_sh  3,32,32,8,102,16,PATTERN_LPS  стало равномерно заполняться!

board707, 2025-09-02:
Отлично!

board707, 2025-09-02:
Проблема решена? Закрываем?